spatialdude/usqlite #2: Build failure esp32

palmtreefrb, 2021-11-09:
First, my apologies for the verbose post(complete build output in zip file). As you can see below the build fails for esp32. Any direction will be appreciated.
Thanks,
Rick

fredrick@fredrick-XPS-13-9370:~/micropython/ports/esp32$ make submodules
git submodule update --init ../../lib/berkeley-db-1.xx
fredrick@fredrick-XPS-13-9370:~/micropython/ports/esp32$ make clean
idf.py -D MICROPY_BOARD=GENERIC -B build-GENERIC  fullclean
Executing action: fullclean
Done
fredrick@fredrick-XPS-13-9370:~/micropython/ports/esp32$ make USER_C_MODULES=/home/fredrick/micropython/modules/usqlite/micropython.cmake

...
[build_failure.zip](https://github.com/spatialdude/usqlite/files/7501461/build_failure.zip)
...
 
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
make: *** [Makefile:34: all] Error 2
fredrick@fredrick-XPS-13-9370:~/micropython/ports/esp32$

palmtreefrb, 2021-11-10:
Ok, I found the issue. I had to add compiler option -mtext-section-literals to the toolchain-esp32.cmake file in esp-idf.
After all this it seems usqlite to too big for my esp32-wroom-32d :-(

bootloader  @0x001000    21104  (    7568 remaining)
partitions  @0x008000     3072  (    1024 remaining)
**application @0x010000  2015200  ( -442336 remaining)**
ERROR: application overflows allocated space of 1572864 bytes by 442336 bytes
make: *** [Makefile:35: all] Error 1

spatialdude, 2021-11-10:
@palmtreefrb Thanks for your feedback and I'm glad you got it to compile. I had been waiting for my ESP32 device to turn up so I could try the module out on this platform. I should have it in the next week or so. 

In the meantime the one of the major keys to reducing the code footprint is to omit or disable SQLite functionality that you do not need. The configuration is defined in [usqlite_config.h](https://github.com/spatialdude/usqlite/blob/main/usqlite_config.h) In this file you will find the default options I selected. This was a first stab at balancing footprint size with functionality. The footprint size can be further reduced with additional `#define SQLITE_OMIT_xxx` entries. These options are described in the [SQLite compile options reference](https://sqlite.org/compile.html) **Options To Omit Features** section.

I don't yet have a good measure of what the footprint impact is for various SQLite features, this is something I plan to put some effort into and provide some higher level configuration choices.

palmtreefrb, 2021-11-11:
That might be tough to eliminate 412496 bytes plus whatever implementation code for running an application. Most ESP32's have 4MB flash but there is a ESP32-WROVER with 8MB flash but it is hard to come by now on a development board. Chip shortages you know.

spatialdude, 2021-11-27:
* Updated [micropython.cmake](https://github.com/spatialdude/usqlite/blob/f990cd74ec001f8c3321d66fe107e8b6065ffdfd/micropython.cmake#L17) to add compiler options for ESP32
* Added [ESP32](https://github.com/spatialdude/usqlite/wiki/esp32) wiki page

@palmtreefrb the change to `micropython.cmake` should avoid having to make any changes to `toolchain-esp32.cmake` in **esp-idf**.

I've got my [TinyPICO](https://www.tinypico.com/) up and running and so far the `usqlite` module is working nicely. (Note I did have to change [`partitions.csv`](https://github.com/spatialdude/usqlite/wiki/esp32#partitionscsv) to accommodate the firmware image's increase in size.

palmtreefrb, 2021-11-27:
Very nice, I will give it a try and let you know how it works for me.

palmtreefrb, 2021-11-27:
Very good, looks great...

build-GENERIC/micropython.bin
or run 'idf.py -p (PORT) flash'
bootloader  @0x001000    21104  (    7568 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  2029360  (  329936 remaining)
total                  2094896

>>> import usqlite
>>> if not usqlite.mem_status():
         usqlite.mem_status(True)
True
>>> con = usqlite.connect("data.db")
>>> con.executemany(
     "BEGIN TRANSACTION;"
         "CREATE TABLE IF NOT EXISTS data (name TEXT, year INT);"+
             "INSERT INTO data VALUES ('Larry', 1902);"+
                 "INSERT INTO data VALUES ('Curly', 1903);"+
                     "INSERT INTO data VALUES ('Moe', 1897);"+
                         "INSERT INTO data VALUES ('Shemp', 1895);"+
                             "COMMIT;")
<Cursor 'NULL'>
>>> with con.execute("SELECT * from data") as cur:
         for row in cur:
                     print("stooge:", row)
                     
stooge: ('Larry', 1902)
stooge: ('Curly', 1903)
stooge: ('Moe', 1897)
stooge: ('Shemp', 1895)
>>> con.close()
>>> print("usqlite mem - current:", usqlite.mem_current(), "peak:", usqlite.mem_peak())
usqlite mem - current: 0 peak: 97296
>>>

spatialdude, 2021-11-28:
Awesome.  Thanks for the feedback @palmtreefrb

closed 2022-01-08 by spatialdude