spatialdude/usqlite #5: compiling don't work with RP4 for the pico

schmesb, 2022-01-06:
Hi,

i want to compile micropython for pico. but the last step don't work. Have anyone a idea why it don't work?

make USER_C_MODULES=/home/pi/my_mpModules/micropython.cmake
[ -d build-PICO ] || cmake -S . -B build-PICO -DPICO_BUILD_DOCS=0 -DMICROPY_BOARD=PICO -DUSER_C_MODULES=/home/pi/my_mpModules/micropython.cmake
PICO_SDK_PATH is /home/pi/micropython/lib/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
PICO compiler is pico_arm_gcc
CMake Error at /home/pi/micropython/lib/pico-sdk/cmake/preload/toolchains/find_compiler.cmake:28 (message):
  Compiler 'arm-none-eabi-gcc' not found, you can specify search path with
  "PICO_TOOLCHAIN_PATH".
Call Stack (most recent call first):
  /home/pi/micropython/lib/pico-sdk/cmake/preload/toolchains/pico_arm_gcc.cmake:20 (pico_find_compiler)
  /usr/share/cmake-3.18/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:50 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
make: *** [Makefile:18: all] Fehler 1

spatialdude, 2022-01-08:
@schmesb It looks like you have a missing `/` in your `make` command between `my_mp` and `Modules` - `make USER_C_MODULES=/home/pi/my_mpModules/micropython.cmake` -->> `make USER_C_MODULES=/home/pi/my_mp/Modules/micropython.cmake`

If that is not the problem then try the following steps to diagnose the problem -
* Confirm you pico sdk is installed and working and you are able to build one of the examples such as *blink* - https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf
* Confirm you can build MicroPython for rp2 - https://github.com/micropython/micropython/tree/master/ports/rp2
* Confirm you have or created a `micropython.cmake` file in the modules folder - https://github.com/spatialdude/usqlite#project-directory-structure

schmesb, 2022-01-08:
@spatialdude Thank you for your prompt reply. I have not forgotten that / that is correct. I'll go through the other steps.

spatialdude, 2023-01-02:
Latest update + MicroPython 1.19.1 builds as expected on RPI400 system - https://github.com/spatialdude/usqlite/pull/14#issuecomment-1368654511

```
[100%] Linking CXX executable firmware.elf
   text    data     bss     dec     hex filename
 662004     100  202948  865052   d331c /home/pi/upython/micropython/ports/rp2/build-PICO/firmware.elf
[100%] Built target firmware
```

closed 2023-01-02 by spatialdude