spatialdude/usqlite #9: insert failed

tljk, 2022-03-25:
esp32wroom32 micropython1.8
`Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4023ffe7  PS      : 0x00060130  A0      : 0x80166468  A1      : 0x3ffcdf40
A2      : 0x0015122d  A3      : 0x3fff0198  A4      : 0x00000005  A5      : 0x00000001
A6      : 0x00000500  A7      : 0x3ffe4f18  A8      : 0x00000001  A9      : 0x00000001
A10     : 0x00000001  A11     : 0xffeaeddc  A12     : 0x00151225  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x00000544  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0015122d  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc

ELF file SHA256: 79f4c1dccd211abb

Backtrace: 0x4023ffe4:0x3ffcdf40 0x40166465:0x3ffcdf60 0x40166fae:0x3ffce010 0x401676f5:0x3ffce060 0x40162eb8:0x3ffce080 0x4015245d:0x3ffce0a0 0x40158c4d:0x3ffce0c0 0x40158d7a:0x3ffce0e0 0x40161811:0x3ffce100 0x40161b82:0x3ffce130 0x401524a9:0x3ffce160 0x40158c4d:0x3ffce190 0x40158d4e:0x3ffce1b0 0x400d174f:0x3ffce1e0 0x400d1c3d:0x3ffce230 0x400d4529:0x3ffce280 0x401049ea:0x3ffce2d0 0x400e16de:0x3ffce320 0x400e1993:0x3ffce360 0x400e1c6b:0x3ffce3a0 0x400ec1c1:0x3ffce3e0 0x400ef11d:0x3ffce440 0x400fe7b1:0x3ffce510 0x400f841f:0x3ffcecd0 0x400f8597:0x3ffced10 0x400d0819:0x3ffced60 0x400d0e00:0x3ffced90 0x400d0773:0x3ffcede0 0x400d038f:0x3ffcee30 0x401524b6:0x3ffcee80 0x40158c4d:0x3ffceeb0 0x40158d7a:0x3ffceed0 0x40084a69:0x3ffceef0 0x4015253c:0x3ffcef90 0x40158c4d:0x3ffceff0 0x40158d7a:0x3ffcf010 0x40084a69:0x3ffcf030 0x4015253c:0x3ffcf0d0 0x40158c4d:0x3ffcf140 0x40158d7a:0x3ffcf160 0x40084a69:0x3ffcf180 0x4015253c:0x3ffcf220 0x40158c4d:0x3ffcf290 0x40158d7a:0x3ffcf2b0 0x40084a69:0x3ffcf2d0 0x4015253c:0x3ffcf370 0x40158c4d:0x3ffcf3d0 0x40158c76:0x3ffcf3f0 0x401810ff:0x3ffcf410 0x401814bd:0x3ffcf4a0 0x40181525:0x3ffcf4c0 0x4016a360:0x3ffcf4e0`

tljk, 2022-03-25:
code only failed when there is some socket operation

spatialdude, 2023-01-02:
@tljk  Please try the latest version of MicroPython and usqlite - https://github.com/spatialdude/usqlite/issues/12#issuecomment-1368655355