spatialdude/usqlite #12: Compile error

emile-cronje, 2022-11-22:
Good day

I am following the getting started section on https://github.com/spatialdude/usqlite

I get the following error when following the instructions on https://github.com/spatialdude/usqlite/wiki/unix

sqlite/usqlite_module.c:127:35: error: ‘mp_obj_type_t’ {aka ‘const struct _mp_obj_type_t’} has no member named ‘make_new’

Ubuntu 22.04

Regards

l-atome, 2022-12-07:
Perhaps watch my new post ... https://github.com/spatialdude/usqlite/issues/13 ... three days to adapt this beautiful code with new update os

spatialdude, 2023-01-02:
@emile-cronje @l-atome Please try out the latest version of µsqlite module. It has been updated to support the latest version of MicroPython (thanks @dpgeorge) as well as updated to the latest version of SQLite (3.40.1).

closed 2023-12-18 by spatialdude