spatialdude/usqlite #13: Compilation on Ubuntu 20.04

l-atome, 2022-12-07:
Un grand merci !!! for this package ...
We introduce python, tcp, sql but also recurcivité, graph, conquer and divide, dynamic programmic for the new curriculum of informatic for the grade 11 and 12
So, use ESP32 with micropython and sqlite, it's amazing for us !!!!

Three day after, i just goal to compil a firmware with micropython and usqlite ....

The big purpose is :
* use gcc8 and not gcc9
* python 3.8 and not python 3.10
* micropython 1.18 and not mycropython 1.19 
Some changes in new upgrade make impossible the compilation of usqlite

More information (in french) on my website : https://chinginfo.fr/dossier-esp32/?p=c

A big thank you for the porting of sqlite to micropython which will make the field of possibilities immense in our teaching

spatialdude, 2023-01-02:
@l-atome Please note - https://github.com/spatialdude/usqlite/issues/12#issuecomment-1368655355

I've also implemented a solution for the recursion problem - https://github.com/spatialdude/usqlite/commit/0d15c6640ac6a92c86e24863e3c37be98dcbfbc9 - ref https://github.com/spatialdude/usqlite/pull/14#issuecomment-1368654511

l-atome, 2023-01-02:
Thanks for your help

closed 2023-01-02 by l-atome