spatialdude/usqlite #17: Compil for ESP32

l-atome, 2023-01-02:
Hello, I just try the new version of uSqlite for micropython 1.196 ... No problems with the ports "unix" but it fails with the port "ESP32" ...

```
/sauvExt/esp32-1.19-4.4/modules/usqlite/sqlite3.c: In function 'sqlite3RunParser':
/sauvExt/esp32-1.19-4.4/modules/usqlite/sqlite3.c:171920:61: error: 'startedWithOom' undeclared (first use in this function); did you mean 'searchWith'?
     assert( db->mallocFailed==0 || pParse->rc!=SQLITE_OK || startedWithOom );

```
It's seems to be a problem of NDEBUG et SQLITE_DEBUG but i'm not able to correct it ...
Thanks for your work
Thomas

spatialdude, 2023-01-02:
@l-atome Using the latest usqlite and MicroPython code I am not able to reproduce the problem you describe. 😕

I am using the steps described here - https://github.com/spatialdude/usqlite/wiki/esp32 (I have a TinyPICO).

Some details that may help you -
I am using Expressif IDF version 4.2 with Python 3.10 running on Ubuntu 22.04.1 in WSL.

```
wsl --version
WSL version: 1.0.3.0
Kernel version: 5.15.79.1
WSLg version: 1.0.47
MSRDC version: 1.2.3575
Direct3D version: 1.606.4
DXCore version: 10.0.25131.1002-220531-1700.rs-onecore-base2-hyp
Windows version: 10.0.22621.963
```

```
lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.1 LTS
Release:        22.04
Codename:       jammy
```

```
Running cmake in directory ~/mpy/micropython/ports/esp32/build-UM_TINYPICO
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DMICROPY_BOARD=UM_TINYPICO -DUSER_C_MODULES=~/mpy/modules/micropython.cmake -DCCACHE_ENABLE=0 ~/mpy/micropython/ports/esp32"...
-- Found Git: /usr/bin/git (found version "2.34.1")
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: ~/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc

...
...

Project build complete. To flash, run this command:
~/.espressif/python_env/idf4.2_py3.10_env/bin/python ../../../../expressif/esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x1000 build-UM_TINYPICO/bootloader/bootloader.bin 0x8000 build-UM_TINYPICO/partition_table/partition-table.bin 0x10000 build-UM_TINYPICO/micropython.bin
or run 'idf.py -p (PORT) flash'
bootloader  @0x001000    23152  (    5520 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  2098432  (  260864 remaining)
total                  2163968
```

l-atome, 2023-01-02:
Hello, i try to compil with 4.2, 4.3, 4.4 and still have an error :
* with 4.3, 4.4 (to include esp32c3 boards), it's a different erreur (see my last post)
* with 4.2, i have this error
```
/sauvExt/esp32-1.19-4.2/usqlite/usqlite_utils.c:101:49: error: 'MP_TYPE_FLAG_ITER_IS_ITERNEXT' undeclared (first use in this function); did you mean 'MP_TYPE_FLAG_EQ_HAS_NEQ_TEST'?
 #define TYPE_HAS_ITERNEXT(type) (type->flags & (MP_TYPE_FLAG_ITER_IS_ITERNEXT | MP_TYPE_FLAG_ITER_IS_CUSTOM | MP_TYPE_FLAG_ITER_IS_STREAM))
```

I try with python3.8 and python3.10 ... And with gcc-8 and gcc-9

This is the code i use to compil usqlite : 
```
git clone -b v1.19 --recurse-submodules https://github.com/micropython/micropython.git ;
git clone -b v4.2 --recursive https://github.com/espressif/esp-idf.git ;
git clone https://github.com/spatialdude/usqlite.git
cd esp-idf
./install.sh
source export.sh
cd ../micropython/mpy-cross
make submodules
make
cd ../ports/esp32
make BOARD= USER_C_MODULES=../../../usqlite/micropython.cmake
```
thanks for your help
Thomas

spatialdude, 2023-01-02:
@l-atome I'm sorry but I cannot reproduce your errors, even with `BUILD=` as the target - see below. The errors you describe appear to be due to constants not found in the MicroPython's [`py/obj.h`](https://github.com/micropython/micropython/blob/master/py/obj.h#L541) which indicates a possible configuration/code version mismatch. 

As you are using a different project directory structure, I would suggest you try organizing your project as described here - https://github.com/spatialdude/usqlite#project-directory-structure and follow the steps without specifying branches for MicroPython and usqlite as that will ensure you are using the latest MicroPython (1.19.1) and usqlite (0.1.3) code.

Then try first try to build for the TinyPICO target as described here - https://github.com/spatialdude/usqlite/wiki/esp32 before your required target.

```
~/mpy/micropython/ports/esp32$ idf.py --version
ESP-IDF v4.2

make BOARD= USER_C_MODULES=~/mpy/modules/usqlite/micropython.cmake
...
...
[ 99%] Building C object CMakeFiles/micropython.elf.dir/project_elf_src.c.obj
[100%] Building C object CMakeFiles/micropython.elf.dir/~/mpy/modules/usqlite/usqlite_module.c.obj
[100%] Building C object CMakeFiles/micropython.elf.dir/~mpy/modules/usqlite/usqlite_connection.c.obj
[100%] Building C object CMakeFiles/micropython.elf.dir/~/mpy/modules/usqlite/usqlite_cursor.c.obj
[100%] Building C object CMakeFiles/micropython.elf.dir/~/mpy/modules/usqlite/usqlite_row.c.obj
[100%] Building C object CMakeFiles/micropython.elf.dir/~/mpy/modules/usqlite/usqlite_file.c.obj
[100%] Building C object CMakeFiles/micropython.elf.dir/~/mpy/modules/usqlite/usqlite_mem.c.obj
[100%] Building C object CMakeFiles/micropython.elf.dir/~/mpy/modules/usqlite/usqlite_vfs.c.obj
[100%] Building C object CMakeFiles/micropython.elf.dir/~/mpy/modules/usqlite/usqlite_utils.c.obj
[100%] Building C object CMakeFiles/micropython.elf.dir/~/mpy/modules/usqlite/usqlite.c.obj

...
...

Project build complete. To flash, run this command:
~/.espressif/python_env/idf4.2_py3.10_env/bin/python ../../../../expressif/esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build-/bootloader/bootloader.bin 0x8000 build-/partition_table/partition-table.bin 0x10000 build-/micropython.bin
or run 'idf.py -p (PORT) flash'
bootloader  @0x001000    21104  (    7568 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  2089184  (  270112 remaining)
total                  2154720
```

l-atome, 2023-01-04:
Hello and thanks for your response and help ... and to @dpgeorge

With the latest micropython + esp4.2 ... It's works and I can continue my project for esp32 ...
I'm new in mycropython and esp32 ...
Have you a forum thread thant can explain me why, it was not work with :
git clone -b v1.19 --recurse-submodules https://github.com/micropython/micropython.git ??

I try with the combo "latest micropython" + esp4.4 and I still get the error :
```
/sauvExt/esp32-1.19-4.4/modules/usqlite/sqlite3.c: In function 'sqlite3PcacheFetchFinish':
/sauvExt/esp32-1.19-4.4/modules/usqlite/sqlite3.c:52852:11: error: implicit declaration of function 'sqlite3PcachePageSanity'; did you mean 'sqlite3PcachePagecount'? [-Werror=implicit-function-declaration]
   assert( sqlite3PcachePageSanity(pPgHdr) );
```
I will try to fix it ... Have you an idea ?? I want build firmware for esp32c3

Thanks agin

gitslav, 2023-03-07:
@spatialdude do you know what changes are required for this library module to work with ESP-IDF 4.4?

l-atome, 2023-03-20:
Hello, for the moment, i did not have time to progress on the production of firmware for ESP32-C3 and ESP32-S3 ... Probably, i'll have more time at the beginning of June .... Thanks .... Thomas