spatialdude/usqlite #21: Micropython 1.21 windows port: Compile error

emile-cronje, 2023-12-11:
Good day

Are there any plans to make usqlite compatible with micropython 1.21? I get a compile error in usqlite_cursor.c in function:

STATIC mp_obj_t row_tuple(usqlite_cursor_t *cursor) {
    int columns = sqlite3_data_count(cursor->stmt);

    mp_obj_tuple_t *o = m_new_obj_var(mp_obj_tuple_t, mp_obj_t, columns);
    o->base.type = &mp_type_tuple;
    o->len = columns;

    for (int i = 0; i < columns; i++)
    {
        o->items[i] = usqlite_column_value(cursor->stmt, i);
    }

    return MP_OBJ_FROM_PTR(o);
}

syntax error: ')' in line:
    mp_obj_tuple_t *o = m_new_obj_var(mp_obj_tuple_t, mp_obj_t, columns);

Regards

spatialdude, 2023-12-18:
@emile-cronje @beyonlo Please try the latest v0.1.4 code, it should resolve the issues with the latest version of micropython.

spatialdude, 2023-12-18:
@beyonlo Yes it does work on microcontrollers and the ones I have personally used it on are included here https://github.com/spatialdude/usqlite#ports Do consider if you have enough memory available on your target device to meet your requirements. The memory usage can also be configured - https://github.com/spatialdude/usqlite#memory-allocation-configuration

spatialdude, 2024-06-01:
@beyonlo There is no official relationship between μSQLite and MicroPython, they're just *friends* 😉

I would be pleased for μSQLite to be included as a module in MicroPython, but the decision would beed to be made by the MicroPython team @dpgeorge on that if there was sufficient demand from the MP community. 

Otherwise I try to routinely keep μSQLite up to date with both MicroPython and SQLite releases.

dpgeorge, 2024-06-02:
I really like this project, and it's good to see @spatialdude keep it up-to-date.

> I would be pleased for μSQLite to be included as a module in MicroPython, but the decision would beed to be made by the MicroPython team @dpgeorge on that if there was sufficient demand from the MP community.

At this stage I still think it makes sense to have this as a separate project, similar to `ulab` https://github.com/v923z/micropython-ulab.

Although we have had requests to bundle `sqlite3` support with the new webassembly port, so I'll be looking into that in the near future.  Will be interesting to see how we can make that work.